ElectroBlocks/ElectroBlocks-Legacy #31: Integrating arduino-cli for local compilation

philippejadin, 2019-01-03:
Hello,

Would you be interested in integrating arduino-cli inside the electron app to allow locale compile as well? I'm progressing albeit slowly on a working solution here  :  https://github.com/philippejadin/blockly-arduino/blob/master/renderer.js

... and I'm tempted to start over and work on your project instead. 

I'm also wondering, what base are you using for the arduino blockly version? BlocklyDuino ? Or something you made on your own?

Thank you anyway!

phptuts, 2019-01-03:
This started off with blocklyDuino, but i am replacing all the blocks right now.  I love the idea, I tried to do something like that myself but I could not get all the OS to work right together.    

I think it would be great if we could pull in all the Arduino Libraries so that the user did not have to install each of them with the Arduino IDE.  I think that's one of the big advantages of compiling online along with not dealing with all the weird OS issues.  But not having to worry about the internet would be huge!  I had situations where I had to use my phone to get things to compile.

Also here is the new blocks, I am working on the generators for them right now.
https://phptuts.github.io/blockly/blockly/demos/code/index.html

Let me know what you think.

phptuts, 2019-04-29:
I thinking ages 10 to 14   That sounds like a fun sensor to add, it's 25 bucks I will get one and start trying it out.  

I am definitely down for a common format.  We should post something on the Blockly google groups.

Sorry the late response my github notification were not setup right.